jtanx/picopterx #13: Picopter-Web Interface

SinaKK, 2015-02-02:
Been reviewing how the web app communicates with the picopter, thinking of starting to port that over (I'll do it in another branch).

I'm happy to continue using thrift unless you have another suggestion?

jtanx, 2015-02-02:
Thrift is fine. The only other one I know of is ZeroMQ which is somewhat different.

SinaKK, 2015-02-04:
The (old) thrift server currently uses boost::thread to launch the different picopter applications and uses timed_join() to check whether another application is currently running.

In changing to c++11, I think it's better to switch to using std::async, std::future and wait_for() [as described [here](http://stackoverflow.com/a/17204246)]

jtanx, 2015-02-04:
If you see the code I've written, I'm already using C++11 threading. No boost dependency right now. I'm a bit short on time this week and possibly next week, which is why I've slowed down a bit. I was planning on doing the lawnmower code next.

I'm a bit dubious of the method used for running various programs and the use of the dodgy `state` variable which determines where it is currently at. At the very least, it should be wrapped in `std::atomic`.

SinaKK, 2015-02-04:
Yup, been following all your code. 

Had the exact thoughts about the 'state' variable... and all these other global variables too.

jtanx, 2015-02-04:
I've never tried, but technically you should be able to compile it directly on OS X too. You'd need to get cmake and co. though. Homebrew for some gpsd?

I've mostly actually been compiling through mingw-w64 (Windows), but as the code becomes more specific, I've been shifting towards a Linux VM. Otherwise compiling on the RPi is too slow in that we're not really writing RPi specific code as of yet, so I might as well use a Linux VM for now.

SinaKK, 2015-02-04:
FYI: unfortunately seems like thrift requires boost.

SinaKK, 2015-02-05:
Contemplating whether to put the generated thrift library in `third_party/thrift` (since it's autogenerated code) or `src/webInterface` (since it's specific to our project)

SinaKK, 2015-02-05:
OK incorporated thrift into CMake and stuff (I'm a newbie at CMake, so feel free to review  and improve it, atm it successfully generates the picopter thrift library code, builds and adds it).  Haven't done any server/client code yet. Also haven't reviewed the .thrift file (protocol) for changes/improvements yet.

jtanx, 2015-02-05:
Looks good! I think it should probably stay in the `third_party/thrift` folder. 

About the boost requirement, that's okay I guess. I think the thrift code that we have to write only makes use of boost's `shared_ptr`, so we could use boost for that and C++11 for everything else. I'll figure out how to link to boost, it [shouldn't be too hard](http://www.cmake.org/cmake/help/v3.0/module/FindBoost.html) (not sure if it works on cmake < 3 though). 

Just something to note about naming conventions, generally for library names, executables, filenames and folders I've been using `name_here` (e.g. lowercase with underscores) instead of camelcase (POSIX convention? De-facto convention? I dunno. Most *nixy stuff remains lowercase).

jtanx, 2015-04-24:
It works now

closed 2015-04-24 by jtanx