jtanx/picopterx #35: Camera tracking: Biasing elevation based on pitch

jtanx, 2015-04-23:
Pitch of the camera means that you have to move too close to the copter before it moves back.

BrettRD, 2015-05-03:
We should be linearising the inputs to our control system by computing an approximate location of the target given the object location in the image and the camera angle.  Either in global or stabilised-body coordinates.
Assuming the camera is stabilised S.T the image X axis is level and the target is on flat, level terrain:
The angle from the image-plane's normal is atan(k*Py) where Py is the number of pixels from centre, k is (40mm/(750mm*Pcal)), Pcal is the number of pixels wide the 40mm calibration square appears from 750mm.
Add the gimbal tilt to this (look up table from a bench test) to calculate a target angle: theta=(gimbal elevation from downward vertical axis)+ (atan(k*Py))
The forward displacement of the target is altitude*tan(theta)
The lateral displacement of the target is altitude*(k*Px)/cos(theta)
Using this, we should be able to track the target with the camera to keep the target in view without disrupting the navigation control loop.
We can take out the flat terrain assumption by adding the Lidar-lite to the camera's centre and finding distance directly once the camera has a lock.

jtanx, 2015-05-19:
As of da188d65016d86a89e6f0d732d, it's possible to use the estimated object height, but I haven't completed the switchover in the object tracker because I don't know how reliable it is. So currently it's still hardcoded to 4m.

jtanx, 2015-09-14:
@BrettRD  is this still relevant?

BrettRD, 2015-09-29:
Hah! gone.

closed 2015-09-29 by BrettRD