jtanx/picopterx #40: MAVLINK

jtanx, 2015-04-28:
We need it

BrettRD, 2015-05-15:
Link Dump:

Basic tute to building the C includes
  http://qgroundcontrol.org/mavlink/create_new_mavlink_message
C bindings exist
  https://github.com/mavlink/mavlink
C uart example
  https://github.com/mavlink/c_uart_interface_example
Python CLI interface:
  http://tridge.github.io/MAVProxy/
ROS driver exitss
  https://github.com/mavlink/mavros
Basic API (using mavproxy maybe)
  http://dev.ardupilot.com/wiki/droneapi-tutorial/
Probably useless
  https://github.com/hybridgroup/gobot/blob/master/examples/mavlink.go

jtanx, 2015-05-15:
We have a Pixhawk (but no GPS) as of 14/05/2015

jtanx, 2015-05-20:
Probably: MAV_CMD_NAV_GUIDED_ENABLE

closed 2015-07-31 by jtanx