jtanx/picopterx #61: Figure out what's wrong with the serial communication to the Pixhawk

jtanx, 2015-06-10:
mavproxy works...

jtanx, 2015-06-16:
Some useful commands:

~~~~~
./ArduCopter.elf --home -35,149,584,270 --model quad
socat  pty,link=/dev/virtualcom0,raw,b57600  tcp:localhost:5760
~~~~~

jtanx, 2015-06-17:
See 67fbb8a27e29c471742e398b6b0223e4babaf020

jtanx, 2015-06-18:
It works if you connect the usb port from the pixhawk to the RPi...

BrettRD, 2015-06-23:
Did we take the Raspian debug terminal off the hardware serial since the switch to Pi V2?

jtanx, 2015-06-23:
Yep, it had to be off so we could use the NAZA gps

jtanx, 2015-06-25:
Seemingly relevant: http://ardupilot.com/forum/viewtopic.php?f=21&t=12776&p=34582

jtanx, 2015-06-25:
* If you haven't connected before: you cannot connect via USB if the system is armed (safety switch 'enabled')
* Once connected via USB you can connect irregardless of safety switch
* For some reason, the native serial port refuses to switch to 57600, both via our code and wiringSerial
* Switching telem2 to be at 115200 allows it to work via RPi **so long** as it comes up first, before the telem1 port.

jtanx, 2015-06-25:
At 2Hz, the mavlink connection currently consumes ~1.3% CPU power. At 5Hz, it's roughly 3%. This with debug messages enabled. Disabled, it's ~2% at 5Hz and ~4% at 10Hz.

jtanx, 2015-06-25:
Message list: https://pixhawk.ethz.ch/mavlink/

jtanx, 2015-07-27:
Haha... wait what...

This is for guided mode:

~~~~~
picopter[2520]: DEBUG: Heatbeat! Mode: 4, 1
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 4, 1
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 4, 1
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
picopter[2520]: DEBUG: Heatbeat! Mode: 0, 0
~~~~~

Mode 4 is meant to indicate guided mode. I have no idea why it only sends 'guided mode' every 1 in X heartbeats.

jtanx, 2015-07-27:
Egh, it's the GCS heartbeats, presumably because I have Mission Planner and MAVProxy connected at the same time...

Good reference: https://github.com/diydrones/MissionPlanner/blob/af52ca40726cfc3b7286256082fb1752ec968853/CurrentState.cs#L974

jtanx, 2015-08-25:
Well it's still somewhat unreliable. Sometimes the serial link will fail (select error; not receiving data from Pixhawk) and the only solution is a full reboot. Other  times the whole RPi will simply not respond to connecting via SSH. It would be interesting to see what happens in this case while hooking up a screen.

jtanx, 2015-09-27:
Closing in favour of #113

closed 2015-09-27 by jtanx