jtanx/picopterx #67: Wireless dongle is broken

jtanx, 2015-07-31:
So it was part software issue (after 1st upgrade attempt, filesystem was corrupt; causing wlan0 to keep losing its IP address) and mostly a hardware issue:

- It looks like the cable in the RP-SMA antenna is severed/on the way to being severed
    - Replaced with my own antenna, *but*:
- The wireless signal tends to drop out - fiddling with the antenna by moving it around brings it back up.

jtanx, 2015-07-31:
I presume we could fix it by opening up the dongle and re-soldering the connections to the antenna...

jtanx, 2015-08-05:
@BrettRD got the wireless router working on it for now (hack)

jtanx, 2015-08-23:
It seems that any dongle with the [RTL8188CUS](http://elinux.org/RPi_USB_Wi-Fi_Adapters) chip is ok, notwithstanding power issues.

The cheapest is [$3.79](http://www.ebay.com.au/itm/High-Speed-Realtek-RTL8188cus-USB-150M-150Mbps-n-Wireless-WiFi-adapter-Lan-Card-/181651117876?hash=item2a4b401734), but if you want one with an external, detachable antenna, it's about [$12](http://www.ebay.com.au/itm/150Mbps-USB-2-0-WiFi-RTL8188-2-4G-Wireless-Internet-Adapter-802-11N-with-Antenna-/271796628652?hash=item3f48576cac) (or [here](http://www.aliexpress.com/item/Mini-150M-Raspberry-PI-2-WiFi-USB-Adapter-Dongle-with-antenna-RTL8188CUS-wireless-network-lan-adapter/32341569431.html))

jtanx, 2015-09-15:
No idea if it's hostapd compatible: http://www.ebay.com.au/itm/Long-Range-USB-WiFi-Adapter-150Mb-6dBi-High-Gain-For-Desktop-Laptop-Raspberry-Pi-/271514118991?hash=item3f3780ab4f

BrettRD, 2015-09-21:
Wow, we really blew this milestone.

jtanx, 2015-09-22:
No, I don't think so. We just need to find a USB dongle that we know is going to work.

I was waiting for a reply on whether or not [this](http://www.ebay.com.au/itm/141478347005?euid=fca369f5154c4b679c09200a760794ae&cp=1&exe=12809&ext=32583&sojTags=exe=exe,ext=ext) was the RTL8188SU chipset (not hostapd compatbile out of the box) or the RTL8188CUS chipset - but I haven't got a reply yet. The other issue is that one is pretty fat so it would occupy the space of two usb ports, but I guess we could always use an extension lead if needed

jtanx, 2015-11-09:
Looks like this will do: Tenda W311M USB Upto 150M Wireless-N Network Adapter

[$8](http://staticice.com.au/cgi-bin/search.cgi?q=wireless%20state%3AWA) *@*MSY