jtanx/picopterx #83: MAVlink Gimbal Data

BrettRD, 2015-08-31:
The object tracker needs to know (or control) the current orientation of the gimbal.

Ideally, we'd still be able to adjust the pose of the camera with the transmitter.

jtanx, 2015-09-06:
@BrettRD, it builds fine on my own computer, so not exactly sure why [Travis is failing](https://travis-ci.org/jtanx/picopterx) with:

~~~~~
/home/travis/build/jtanx/picopterx/code/src/modules/object_tracker.cpp: In member function ‘void picopter::ObjectTracker::EstimatePositionFromImageCoords(picopter::GPSData*, picopter::FlightData*, picopter::IMUData*, picopter::ObjectInfo*)’:

/home/travis/build/jtanx/picopterx/code/src/modules/object_tracker.cpp:321:33: error: conversion from ‘cv::Matx<double, 3, 1>’ to non-scalar type ‘cv::Vec3d {aka cv::Vec<double, 3>}’ requested

     cv::Vec3d RelBody = Rgimbal * RelCam;

                                 ^

make[2]: *** [src/modules/CMakeFiles/picopter_modules.dir/object_tracker.cpp.o] Error 1

make[1]: *** [src/modules/CMakeFiles/picopter_modules.dir/all] Error 2

make: *** [all] Error 2
~~~~~

jtanx, 2015-09-06:
Never mind, added some random PPA to get OpenCV 2.4 and it's fine now.

jtanx, 2015-09-08:
With the FlightBoard overhaul in progress, I've confirmed that the Pixhawk does send where it thinks the gimbal is via the ArduCopter specific [MOUNT_STATUS](https://github.com/jtanx/picopterx/blob/fb-overhaul/code/src/base/flightboard.cpp#L158) message.

Interestingly, and unless I've decoded the message wrong, it seems that it considers our roll to be our pitch and our pitch to be our roll.

jtanx, 2015-09-08:
Also: The simulator does not send these MOUNT_STATUS messages.

jtanx, 2015-09-08:
@BrettRD I've merged the latest master (with all the flightboard changes) into your observations branch.

Since the `SetBodyVel` command controls altitude with the z parameter, there is now a separate function `SetYaw` to control yaw. I'm not sure if you still want to use relative commands for this purpose - I don't know if using `SetGuidedWaypoint` would be better or not.

jtanx, 2015-09-08:
@BrettRD  I've just merged what is currently in your observations branch into master, because it actually has at least a semi-working object tracker vs. something that will likely make the copter behave erratically if we do object tracking. You can still continue working on the observations branch though.

BrettRD, 2015-09-27:
I've zeroed the mavlink gimbal messages using down as the neutral position.
The servo can reach just past horizontal (0deg) and vertical (+90deg).
I've used the follwing settings in mavlink.
Pitch:
reverse,
Min = 1006uS, -10deg
Max = 2100uS, +90deg
Roll:
reverse,
Min = 1100uS, -55deg
Max = 1900uS, 40deg

This will apply until the camera falls off again.