jtanx/picopterx #92: Tent detection

jtanx, 2015-09-14:
Detect where the tent is...

jtanx, 2015-09-15:
@manishmohanty09 

* From the web interface - set camera to threshold for white objects
* Modify your code to see if an object has been detected in the frame - if so, then it's detected a white object. Assume it's the tent.
* Use the position information, and perhaps @BrettRD's code to position the copter relative to the tent

BrettRD, 2015-09-21:
The observation code is coming together in the object tracker.
We can probably leave the object tracker running in observation mode and share the list of trackable things.
It's probably enough to just build a new trajectory algorithm for the object tracker