jtanx/picopterx #107: Add public home coordinate parameter to FlightController class

20739367, 2015-09-23:
GridSpace class needs to use the coordinate of the home location to create the collision grid world. Also a generally useful piece of information to have access to for navigation purposes.

jtanx, 2015-09-23:
Haha... looks like it's not implemented in ArduCopter (`MAV_CMD_GET_HOME_POSITION`)

jtanx, 2015-09-23:
One workaround is to monitor when the system goes from `disarmed` -> `armed` state and record the current GPS postiion. This won't work if the copter's already in the air and you start the server (happens sometimes if we want to recompile/retest something *on the fly*, so to speak).

jtanx, 2015-10-01:
Done.

Call with `fc->fb->GetHomePosition(&home);`

Note it returns `false` if we don't have a home position. So probably wise to check the return value and to stop doing anything if you don't know what the home position is.

closed 2015-10-01 by jtanx