jtanx/picopterx #108: coord3D to NED

20739367, 2015-09-23:
Need function to take 2 Coord3Ds in GPS coordinates and find the NED vector between them

jtanx, 2015-09-23:
There's already CoordDistance and CoordBearing - I guess you could extend this to some convenience function to calculate the Cartesian form from this.

BrettRD, 2015-09-24:
My latest commit has this in the object tracker.
We may like to migrate these into the navigation header.

jtanx, 2015-09-24:
Do you mean [these](https://github.com/jtanx/picopterx/blob/observations/code/src/modules/object_tracker.cpp#L363) functions?

BrettRD, 2015-09-24:
Whoops!  Forgot to push.
[These ones.](https://github.com/jtanx/picopterx/blob/observations/code/src/modules/object_tracker.cpp#L581)

jtanx, 2015-09-24:
I guess a simple option using current functions may be

````c++
/**
 * Return the distance (in m) between two coordinates.
 * @param c1 The first coordinate (from)
 * @param c2 The second coordinate (to)
 * @return The NED distance (in m).
 */
Vect3D CoordDistance(Coord3D c1, Coord3D c2) {
    double bearing = CoordBearingX(c1, c2);
    double radius = CoordDistance(c1, c2);
    return Vect3D{radius*sin(bearing), radius*cos(bearing), c2.alt - c1.alt};
}
````

Second version
````c++
/**
 * Return the distance (in m) between two coordinates.
 * @param c1 The first coordinate (from)
 * @param c2 The second coordinate (to)
 * @return The NED distance (in m).
 */
Vect3D CoordDistance(Coord3D c1, Coord3D c2) {
    double x = DEG2RAD(c2.lat-c1.lat)*1000*RADIUS_OF_EARTH;
    double y = DEG2RAD(c2.lon-c1.lon)*1000*RADIUS_OF_EARTH*DEG2RAD(c1.lat);
    return Vect3D{x,y, c2.alt-c1.alt};
}
````

Untested.

BrettRD, 2015-09-24:
I'm leaning quite heavilly on the openCV matrix operators and I don't think I'll be getting away from them any time soon.
I've changed my ground coordinates to cv::Vec3d type.