jtanx/picopterx #113: RPi sometimes crashes

jtanx, 2015-09-27:
The RPi on the Copter will sometimes hard freeze. This happens both when using the battery and using the plug pack. It's really annoying me (plus somewhat concerning)...

It's either:
* Disk corruption (try another SD card?) We should make a backup image this week.
* Power issues - maybe with the regulator. Should measure the output with a CRO...
* Dodgy RPi 
* Somethinge else

BrettRD, 2015-09-29:
Tests:
- [ ] Disk corruption: Pull the disk, backup and scan.
- [ ] Brown-outs: Is there a brown-out flag available on boot? most microcontrollers have this.
- [ ] Dodgy RPi: I hope not.
- [ ] Somethinge else: Everything else.

jtanx, 2015-09-29:
Brown-outs: https://www.raspberrypi.org/forums/viewtopic.php?t=82373

jtanx, 2015-09-29:
What I do notice when it crashes both LEDs (green and red) on the RPi glow solid green.

jtanx, 2015-10-08:
@BrettRD I noticed the power warning today even while powered of the plug pack so I'm leaning towards a power issue.