jtanx/picopterx #119: Integration of Grispace into main loop

20739367, 2015-10-01:
Picopter needs to create a Gridspace object, associate it with the existing PathPlan object, and call the raycast() function a few times every second as it flies.

jtanx, 2015-10-02:
Currently one PathPlan object is constructed each time the waypoints module is started. How would this be handled with the Gradspace object in place too?

jtanx, 2015-10-02:
@20739367 the pathplan system is working well though! One thing it doesn't like is if it is given waypoints that are inside exclusion zones, e.g:
[image]

In this case it loops forever and eventually exhausts memory.

20739367, 2015-10-02:
I'll see what I can do about invalid waypoints. 

The gridspace must be persistent, as it stores the copter's observations of the environment. I had thought the path plan was persistent as well. Currently the girdspace is dependent on Pathplan and creates polygons as obstacles are observed in the environment.

If it's essential that we destroy and reconstruct path plans as we go, I could do something like have the pathplan read the grid space and create polygons in its' constructor. 

What do you think?

jtanx, 2015-10-09:
Just another bug which may be related:
~~~~
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffda6ac700 (LWP 2180)]
0x000000000057e1de in picopter::PathPlan::addPathEdge (this=0x7fffda6ab110, 
    n1=0, n2=4) at /home/jeremy/fyp/picopterx/code/src/modules/pathplan.cpp:164
164	    paths[n2][n1] = length;
~~~~~

Paths was of size 4, `n2=4`, `n1=0` (the access to 4 causes the segfault because permissible indices are between 0-3 in this case). I don't understand enough to know why paths was originally of size 64 but its size gets reduced to 4 - perhaps the erase calls?

Anyway, I was using a spiral pattern with the following waypoints:
1. Lat: -31.9799003938478, Lon: 115.817997157574
2. Lat: -31.9797684355908, Lon: 115.818201005459
3. Lat: -31.980068754107, Lon:115.817342698574

The exclusion zone had the following points (approximately):
1. Lat: -31.979745684148, Lon: 115.817890875042
2.  Lat: -31.9799771798151, Lon: 115.818275436759
3: Lat: -31.9799686480442, Lon: 115.81777151674