jtanx/picopterx #124: Object tracker behaviour confusion and DO_MOUNT_CONTROL

BrettRD, 2015-10-04:
The object tracker is using guided waypoints now, and it's so much better.

The simulator gimbal faces straight down, so anything below centre of the image is behind the copter and it will respond by turning around and backing off. (or just chasing its tail)

I've set demo mode to ignore pitch and roll, so with slope set to 10, you can get it to chase a target at the top of the screen or back away from one just above the centre line.

I've implemented DO_MOUNT_CONTROL in the flight board, but it's only supported by ardupilot 3.4+

jtanx, 2015-10-05:
You can control the mount through the APM specific [`MOUNT_CONTROL`](http://mavlink.org/messages/ardupilotmega#MOUNT_CONTROL) message (which is subsequently deprecated in version that implement DO_MOUNT_CONTROL).

BrettRD, 2015-10-05:
I've implemented the apm style messages now, and they seem to be accepted, but it still doesn't control the gimbal.

jtanx, 2015-10-05:
In the simulator? IIRC the simulator doesn't emulate the gimbal at all.